jon-harper/E34M1 #6: Belt holder a bit too high (Merc One.1, v1.1.3)

k-piekarczyk, 2023-04-19:
Hi @jon-harper!
First of all I love that you adapted EVA3 for Merc One.1, and the fact that you took your time to show some assembly instruction.

The belt grabbers seem to be 2-3 mm to high (at least for my setup), as pictured.

[image]

Would you be willing to share the STEP file for the front mount (the one with holes for ADXL345), so I can fix that for myself? I couldn't find it in the files already in the repo, but maybe I missed it.

jon-harper, 2023-04-20:
First, here's the latest [Core assembly STEP file](https://github.com/jon-harper/E34M1/blob/main/CAD/Core.step). For some reason it wasn't uploaded.

OTOH, you should not have have that angle on the belts. The only time I've seen this is with the early revision front before the belt catches had a divider. I can't see from this angle, but the belt catches should look like this:

[image]

If you have an older Front piece without the divider you can still straighten the belts; it just takes more fussing.

If none of that helps you, can you send pics of both sides of your belt catches?

k-piekarczyk, 2023-04-21:
From what I can see, the front I got from the documentation website (11.04.2023) is different, the new one should fix my issue, I'll find out diring the weekend :)

Disaster1971, 2023-04-23:
Same Issue I'm having with my belts. I think it might be effecting the bltouch mount to were the nozzle sitting higher then the bltouch

jon-harper, 2023-04-23:
At this point I'm considering this an open issue in need of a fix. I'm recovering today from traveling but plan to test a solution tomorrow.

jon-harper, 2023-04-26:
Two of us are now running the modified front from [this commit](https://github.com/jon-harper/E34M1/commit/b297bf8145c11a9171a5439548775daa3bd0f18a) and it looks good. 

The belt catches dropped 1mm. Both experimentation and measurements keep coming up with *exactly* 1mm, so I'm suspecting I made a math error along the way in CAD.

Considering printing the new belt grabbers, as well. They do a better job engaging the belts without bowing.

k-piekarczyk, 2023-04-26:
The v0.1 version of the front works fine for me now, though I must admit that the new belt loop instead of the earlier open back design was inconvenient to mount, since the top of the belt loop drooped a bit, making it impossible to just push the belt through.

[image]

jon-harper, 2023-04-26:
> The v0.1 version of the front works fine for me now, though I must admit that the new belt loop instead of the earlier open back design was inconvenient to mount, since the top of the belt loop drooped a bit, making it impossible to just push the belt through.
> 

I think you may have the wrong file. It sounds like you printed an older test version that I accidentally committed and later replaced. Not sure what happened, but the version I linked should be labeled v0.2. It's the latest `stock_front.stl` in the testing branch if you want to grab it from there.

jon-harper, 2023-05-02:
Several users are running v0.2 now and it's included merged into `main`. :+1:

closed 2023-05-02 by jon-harper