jon-harper/E34M1 #7: Dual 5015 Rear Intake Fit Testing

jon-harper, 2023-04-23:
There's a new, dual-5015 radial fan rear intake. It's modified from a stock EVA 3 "cooling inlet" to E34M1-style with screw holes for the 4x 45mm screws that hold the Core together.

[image]

[image]

## Bill of Materials

| Parts | Qty |
|---|---|
| dual_5015_inlet.stl | 1 |
| 5015 radial fan | 2 |
| M3 x 45 SHCS | 4 |
| M3 x 6mm SHCS | 1 |
| Lock washer, M3 | 5 |
| M3 x 20mm SHCS | 2 |
| Heat set inserts, M3x5x4 | 2 |

sylwester-, 2023-05-02:
printed it yesterday. fits well, works well!

jon-harper, 2023-05-02:
Thanks @sylwester- !

closed 2023-05-02 by jon-harper