jon-harper/E34M1 #11: Sherpa Mini fit test

jon-harper, 2023-07-13:
Sherpa Mini has a mount available, but fit test is pending. See the docs.

sharpner, 2023-07-27:
[image]

I use that mount, the only thing that did not quite work was to get that screw in there - but it works okayish without :-)

jon-harper, 2023-07-27:
> [image]
> 
> I use that mount (https://github.com/jon-harper/E34M1/blob/testing/STL/drives/drive_sherpa_mini.stl), the only thing that did not quite work was to get that screw in there - but it works okayish without :-)

Was the screw hole off or is there not clearance with the ADXL?

sharpner, 2023-07-27:
There is no clearance (without the board of course) - I can remove my board and make a photo later to better show the problem

jon-harper, 2023-08-25:
> There is no clearance (without the board of course) - I can remove my board and make a photo later to better show the problem

Gotcha, I've got it installed now and see the issue. Did you try with a button head screw? I'm using the PIS front and it just fit.

sharpner, 2023-09-01:
[image]
Similar, but I just rasped it of a bit and now I got the screw in there 🤷‍♂️

jon-harper, 2023-09-01:
Do you happen to have a button head screw to test it with? It fits with a button head on mine but I'm using the PIS front.

sharpner, 2023-09-02:
ah sorry, I assumed you meant SHCS because I was using the button head screw initially. As you can see I used it for the other two screws as well. Yes I am using one now, but after I scraped off a bit.

jon-harper, 2023-09-04:
> ah sorry, I assumed you meant SHCS because I was using the button head screw initially. As you can see I used it for the other two screws as well. Yes I am using one now, but after I scraped off a bit.
> 
> So it's no longer an issue for me if you want to close the ticket xD

I made a note about the clearance issue and recommended a button head screw there. I'll push the updated docs soon.

Thanks for your help!