jon-harper/E34M1 #13: EBB42 does not fit

drright72, 2023-09-09:
EBB42 CANbus board does not fit when using dual 5015 fan inlet.

jon-harper, 2023-09-09:
This version is in testing to address that: https://github.com/jon-harper/E34M1/blob/testing/STL/rear/dual_5015_inlet.stl

A user gave a preliminary report that it fits, but hasn't been able to use it. Let me know how it goes so I can merge it into the `main` branch if it works or fix any issues if they occur.

Thanks!

drright72, 2023-09-09:
This barely accommodates the EBB42 board on a 23.5mm stepper, but WITHOUT the CAN board mount. I think making the base 10mm thicker would cover it.

jon-harper, 2023-09-10:
It didn't click that you are using a BMG. Most folks have or are moving away from NEMA17 because of size and weight; the BMG/NEMA17 is the largest drive module. The dual rear inlet is the largest rear module, as well, and has been redesigned twice to reduce the size and get it working with NEMA14-mounted PCBs. There's just very little room to adjust things in there.

My recommendation is to go single rear inlet until you can switch to a NEMA14 and Sherpa Mini (it's printed and uses BMG parts, so no need to buy the extruder). Another option would be a 10mm shim under the drive module that you mentioned, though that comes with its own issues.

drright72, 2023-09-10:
Yes, I'm using the BMG. I do have the parts for Sherpa, and had intended on using it with an EBB36. Unfortunately I decided on the Sherpa Micro before realizing there's no good way to secure the motor like with the Mini (the Micro mount lacks the holes for a zip tie). Too much unsupported mass slinging around for just the two holes on the Sherpa Micro to withstand. I managed to get the offset mount I was using before reprinted on another printer, so I can get the M1 back together and print out the Sherpa Mini parts. The other printer works in a pinch, but it's slow and the parts aren't as strong.

closed 2023-09-10 by drright72