jon-harper/E34M1 #14: Front modification for Ender 6 Usage?

RodentVienna, 2023-09-20:
Hi Jon!

First of all, THANK YOU for this project! I have been looking forever to find EVA3 compatibility for Ende 6. E34M1 is so very close.
I actually would like to ask, if there is a chance you would consider including a variant that directly allows compatibility for Ender 6.

the actual differences for Ender 6:
- Belts have teeth looking to the front, not to the back
- Belt mount compartments with slightly different position and dimensions.

By chance you would not have f3d files for the parts?

Alternatively I work out something myself and post the result as a remix on printables.com, but i thiought it may be  of interest to you, to officially add one more feature to the list - native ender 6 support.

Thank you!
Best regards, Alex

jon-harper, 2023-09-20:
I'm sorry, but making those modifications would be a large project, and I lack access to an Ender 6 for testing. 😢

You can find STEP files in the /CAD folder. Not all of them are entirely up to date, but they will work fine as a starting point for you.

closed 2023-09-20 by jon-harper

RodentVienna, 2023-09-20:
Fair enough. After inspecting my situation further, my entire xy gantry requires rebuild anyway for this to make sense, as the ender 6 has its belts positioned around 30mm lower from the x_2020 profile, so that appears not compensatable even by the cleverest redesign of the FRONT part :)