jon-harper/E34M1 #28: CRTouch UHF Mount

121Jigawatt, 2023-12-17:
The CRTouch mount for both the UHF and the non UHF look to be the same model unless i'm missing something.
Was that a mistake or does it use the same mount?

jon-harper, 2023-12-17:
Ack, this is definitely a mistake. There was an issue with the CR Touch model that needed fixing, but this looks like a simple 'oops'.

I will try to get a fixed UHF mount to you today.

121Jigawatt, 2023-12-17:
Thanks so much! Also I have made an edit that allows for this filament cutter to be used with the rapido on your E34M1 tool head.
https://www.printables.com/model/647352-filament-cutter-for-eva3-orbiter-rapido/files
I'm new to using github so what would be the best way to get the stls to you if you want them?
Our would you rather add a link to the printables page when I upload it?

jon-harper, 2023-12-17:
> Thanks so much! Also I have made an edit that allows for this filament cutter to be used with the rapido on your E34M1 tool head. [...]

That is a really neat design! Unfortunately, the modified Front makes it incompatible with E34M1. :-/

What about a thin spacer between hotend and extruder mount? Is that too high up to cut?

121Jigawatt, 2023-12-17:
It could definitely be done but at the cost of more wasted filament purge :/ 
The link that I sent was what I based it off of I edited your design to make it fit. I did have to remove the top font right extruder mounting hole to make it fit unfortunately. Cuts it right at the top of the heatsink. I getcha though, kinda takes away the whole modular design of the whole system. Thanks for looking at the abl mount though!
[image]

jon-harper, 2023-12-17:
> It could definitely be done but at the cost of more wasted filament purge :/ The link that I sent was what I based it off of I edited your design to make it fit. I did have to remove the top font right extruder mounting hole to make it fit unfortunately. Cuts it right at the top of the heatsink. I getcha though, kinda takes away the whole modular design of the whole system. Thanks for looking at the abl mount though! ![20231217_015811](https://private-user-images.githubusercontent.com/125784340/291051743-c76b7802-6318-4e92-a06f-a8b92c998a63.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI4MDQxMzgsIm5iZiI6MTcwMjgwMzgzOCwicGF0aCI6Ii8xMjU3ODQzNDAvMjkxMDUxNzQzLWM3NmI3ODAyLTYzMTgtNGU5Mi1hMDZmLWE4YjkyYzk5OGE2My5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxN1QwOTAzNThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00OGM0ZTgxM2YxNzcxYjlmZTVlYzAzOWZmY2IwYmUwNjk5YTI2YzkwNjk4ZGYzNzQ0MTY5NDA5ODMzNzQ0NWJkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.TAUqAV19A2O5t4pzhVWri9K-70lJYe1_69RjKjYFgYE)

Oh my, I did not catch that. Color me more interested. If you have Discord, consider dropping by the ZeroG EVA 3 channel and spreading the word. I know *several* users who might be interested. Meanwhile, I'll share your printables link.

121Jigawatt, 2023-12-17:
Will do! I thought that might have been the case lol. Working on a way to actuate the cutting element with something similar to the X axis stop so I won't post anything until I get that worked out hopefully early next week

jon-harper, 2023-12-17:
There's a new `abl_crtouch_uhf.stl` file. Let me know if you run into any other issues.

[image]

And don't be a stranger if you want to punt ideas around! :-)

121Jigawatt, 2023-12-17:
Thanks! What's the best way to contact you, in that discord I'm guessing?
Also what is that bottom design, it looks awesome

jon-harper, 2023-12-17:
> Thanks! What's the best way to contact you, in that discord I'm guessing?

I tend to keep the ZeroG Discord open a lot, yeah.

> Also what is that bottom design, it looks awesome

That's Kraken. https://github.com/jon-harper/E34M1/tree/testing/STL/bottom It's been in testing, but is ready for release. I'm working on the docs this weekend. :-)

closed 2023-12-17 by 121Jigawatt

121Jigawatt, 2023-12-17:
Is it the Toolhead channel or is there a dedicated EVA3 channel that I'm not seeing?

jon-harper, 2023-12-17:
> Is it the Toolhead channel or is there a dedicated EVA3 channel that I'm not seeing?

It's a little hidden. Right here: https://discord.com/channels/747612067951018075/1041406799342997544