jon-harper/E34M1 #29: Bottom horns for Rapido HF vs Rapido UHF

NikolaCv, 2023-12-20:
Greetings!

I've recently noticed a change in the design of the Rapido hotend mount, which now comes in two versions – one for HF and the other for UHF that allows for a slightly higher hotend mount (1.6mm). However, if (picture below) the UHF hotend is only compatible with UHF bottom horns when using the UHF hotend mount (and HF only with HF parts), it means that switching from HF to UHF needs changing both the bottom horns and the hotend mount, instead of just changing the horns. This adds the extra step of unscrewing the hotend itself, making the process more time-consuming.

[image]

Is it possible to revert to a single mount for both Rapido HF and UHF? Or does it even matter, since it's just a 1.6mm difference? I haven't printed the parts yet, so I don't know what's the height clearence from the nozzle tip to the bottom of the horns etc.

Any insights on this would be appreciated!

jon-harper, 2023-12-20:
EVA specifies the height of a nozzle should be, because the ducts have to
point at them. The hotend mount isn't there just to hold the hotend in
place: it also puts it at the right height for those ducts.

The 1.6mm difference between the two mounts is to get the Rapido UHF to the
right height for a UHF hotend.

Hope that explains why the Rapido and Rapido UHF don't share a mount.

- Jon

NikolaCv, 2023-12-20:
>The hotend mount isn't there just to hold the hotend in
place: it also puts it at the right height for those ducts.

Why not simply lower the ducts by 1.6mm instead of introducing a new hotend mount for UHF? Given that UHF requires a new Z offset and ducts anyway (roughly 9mm difference), adjusting the mount by 1.6mm doesn't maintain the nozzle at the same height.

jon-harper, 2023-12-21:
The Rapido UHF mount is the height it is because it conforms to the EVA 3
specification. I can't change this, as I did not create the spec.

NikolaCv, 2023-12-21:
What would be the problem if I used Rapido UHF with HF hotend mount, and modified the UHF ducts to match the nozzle height properly (make them point 1.6mm lower)? I genuinely don't see why this wouldn't work.

jon-harper, 2023-12-21:
It probably would work, but it wouldn't conform to the EVA spec anymore. Also ABL might be an issue if not also adjusted.

closed 2023-12-28 by NikolaCv