jon-harper/E34M1 #37: Dragon UHF "bottom horns" clarification

grilparto, 2024-05-09:
Not sure if this technically fits as an issue, but I wanted to encourage a minor clarification to the Dragon UHF Bottom Horn recommendations in the Intro Guide (https://jon-harper.github.io/E34M1/guide/) and the general notes for the Dragon UHF hotend mount (https://jon-harper.github.io/E34M1/modules/hotend/).

Context:  There are two possible configurations for the Dragon UHF.  One involves threading the nozzle directly into the heat block.  This configuration is not really capable of the "UHF" flow rates, instead it's on par with the HF version of the hot end.  The other configuration is to thread on an extension nut to the heatblock, and then thread the nozzle into the extension nut (Phaetus refers to it on the product page as an "extended nut".  The extension nut acts as an extended melt zone--functionally similar to a volcano hotend--allowing for the increased UHF flow rates.  The Rapido UHF appears to have the same configuration options (the user can choose to forgo the extension nut if they don't need the added flow rate).  Not sure why people seem to be commonly forgoing the extension nut on the Dragon UHF but not the Rapido UHF, but I digress.

The clarification: In any case, the first configuration works with the Dual or Kraken bottom horns as stated in the documentation.  However!  The second configuration does require the UHF or Kraken UHF bottom horns like the Rapido UHF.  I don't have the Rapido UHF to check, but I would assume that the same logic should also apply to the Rapido UHF if the user chooses to remove the extension nut.

jon-harper, 2024-05-14:
Have you tried testing this with the UHF horns to see if it's the correct height with the melt zone extension?

grilparto, 2024-05-14:
> Have you tried testing this with the UHF horns to see if it's the correct height with the melt zone extension?

Visual inspection, it looks like the air would be directed towards the nozzle correctly, but I still haven't done a functional test yet.  The non-UHF would aim at the base of the heater block, the UHF reaches the nozzle.  I can take a photo of it when I get back home.  

Once I finish getting this all wired up and running again, I can report back on a functional test. Is there a specific way that I ought to test this?  I generally just check to see if airflow is at the nozzle tip, but not so low that it significantly disrupts the path of the extruded filament.  I've also tried using a small bowl or dish of water to get an idea of where the air starts to hit vs the nozzle height, but I don't know off-hand if there's a hard/fast rule I should conform to.

jon-harper, 2024-05-23:
The bottom of the horns should be about 1mm above the nozzle. If it's more than that you may run into issues.

grilparto, 2024-05-25:
> The bottom of the horns should be about 1mm above the nozzle. If it's more than that you may run into issues.

Oof!  Sorry, been engrossed with several hobbies this week.  Forgot to post a picture.

Looks like it's about 1-1.5mm above the nozzle tip, so probably ok.  Hard to measure with the calipers or without a feeler gage, but I kind of used a tweezers as a compass and then measured the tweezers gap with a ruler.  I did a quick PLA print yesterday (because I needed a widget for one of my other hobbies) and it seemed to work fine.  It was a pretty tame object though so I'll have to put it through its paces to see if the positioning is adequate for bridging and overhangs.

Here's a pic from a week ago:

[image]

jon-harper, 2024-05-25:
It's hard to judge the height from a picture, but I'll trust your results.
Can you do a benchy and see how the bow looks? Don't rotate the model.

jon-harper, 2024-05-29:
> Just wanted to give a quick follow up. I ran into some delays on this
> while I'm trying to tune everything after re-assembly. Chasing down some
> kind of mechanical issue (per my garbage Y-axis resonance graph). I'll give
> the Benchy a try once I have it printing again.

No worries. I appreciate you following up.

grilparto, 2024-05-30:
Suppose I should mention a couple of confounding factors to preface these photos.  I'm printing with a 0.6mm nozzle, and using a tent enclosure.  Both present some interesting challenges when it comes to printing a Benchy (or anything small).  Which is to say that it's like drawing fine detail with a magic marker and cooling pla with hot air.  Also didn't help that I forgot to reposition the seam to a corner.  In any case, the results pretty much align with what I've had to deal with/workaround in the past.

Layer height was 0.3mm.  For the most part, for the better print, the print speed was around 60mm/s (give or take depending on layer time).  For the "fast" print, it was 100mm/s (again, give or take).  I'm debating about swapping out for the 0.4 nozzle and trying again but I don't generally print with that on the larger Ender 5 Plus anyway.

The main problems encountered are ones I'd generally attribute to not being able to cool the filament fast enough.  Is it the duct's fault?  I'm not overly convinced since I'd been grappling with similar issues with an adjustable height duct previously.  I'm more inclined to believe that the single 5015 fan and the ambient heat in the chamber are mostly responsible.  Additionally, I think the water test results I also included below are convincing enough for me to accept the duct height.  I think the dimple could be _little_ flat which might suggest that the fan is a shade higher than ideal, but it's still hitting the layer being printed and several under it.

[image]
[image]
[image]
[image]
[image]

For the water test below, I leveled the nozzle tip with the water and then adjusted the fan speed for 25%, 50%, 75%, and 100%.

(First one is 0%)
[image]
[image]
[image]
[image]
[image]

jon-harper, 2024-05-31:
I designed the Kraken duct to work around concerns regarding the direction of airflow with the stock dual horns. EVA has an unresolved issue about this: https://github.com/EVA-3D/eva-main/issues/117 I apologize for not mentioning the EVA issue sooner; I just remembered it.

Kraken is not perfect, but it does give more uniform distribution of air than the dual horns. You might try it, even if only for comparative purposes.

PXL_20240530_151948875.RAW-01.COVER.jpg (view on web) <https://github.com/jon-harper/E34M1/assets/82681495/a36e7dc3-7ddf-4c4b-9912-5781d4486db5> 
PXL_20240530_164616238.RAW-01.COVER.jpg (view on web) <https://github.com/jon-harper/E34M1/assets/82681495/9034e676-58c5-4955-b407-440c2b7d0847> 
PXL_20240530_164626281.RAW-01.COVER.jpg (view on web) <https://github.com/jon-harper/E34M1/assets/82681495/71844d02-01a3-442a-b40a-fdf18580f2fe> 
PXL_20240530_164636591.RAW-01.COVER.jpg (view on web) <https://github.com/jon-harper/E34M1/assets/82681495/ae311801-42da-4462-b212-1f9dde48067f> 
PXL_20240530_164653775.RAW-01.COVER.jpg (view on web) <https://github.com/jon-harper/E34M1/assets/82681495/c6bd013c-89b3-4973-bf8a-c424c3ee3713> 

PXL_20240530_180101279.RAW-01.COVER.jpg (view on web) <https://github.com/jon-harper/E34M1/assets/82681495/30d83eb2-cc0a-4b05-9fe6-3ae683fd73bb> 
PXL_20240530_180125951.RAW-01.MP.COVER.jpg (view on web) <https://github.com/jon-harper/E34M1/assets/82681495/d22a6018-ec73-4ce9-9308-0f28f6b1819c> 
PXL_20240530_180223350.RAW-01.MP.COVER.jpg (view on web) <https://github.com/jon-harper/E34M1/assets/82681495/77b136ee-ade9-42ba-a348-76f55804a8ce> 
PXL_20240530_180258103.RAW-01.MP.COVER.jpg (view on web) <https://github.com/jon-harper/E34M1/assets/82681495/4de73240-f60f-4100-8fb3-4e81c5d44feb> 
PXL_20240530_180413166.RAW-01.MP.COVER.jpg (view on web) <https://github.com/jon-harper/E34M1/assets/82681495/50b95fc4-4f2f-4a77-a1d9-75dfc9d838d1>

grilparto, 2024-05-31:
> I designed the Kraken duct to work around concerns regarding the direction of airflow with the stock dual horns. EVA has an unresolved issue about this: [EVA-3D/eva-main#117](https://github.com/EVA-3D/eva-main/issues/117) I apologize for not mentioning the EVA issue sooner; I just remembered it.

No worries!  So, the stock duct with the 5015 fan puts the air in front of the nozzle a little instead of under it?  Interesting.  I can give the Kraken a try.  I just printed a dual 5015 inlet and was also looking at the CPAP one (but I don't think the E5Plus stock PSU can handle the cpap fan's power consumption while the rest of the system is under full load).

jon-harper, 2024-07-23:
How did this turn out for you? Did Kraken work better?