jon-harper/E34M1 #38: top_left_microswitch.stl incompatible with drive_bmg.stl - Not enough clearance for cable anchor

grilparto, 2024-05-10:
When integrating the  top_left_microswitch.stl with drive_bmg.stl, there is not enough clearance to install the cable anchor due to the NEMA17 motor's size.  The only workaround is to remove the cable anchor.  My fusion360 skills are currently awful, but if I manage to come up with a modded design, I'll include it here.

grilparto, 2024-05-10:
Came up with an offset version that should fit with the BMG/NEMA17 configuration.  It fits around the NEMA17 and I also added a boot heel (draft, or whatever it's technically called) so that the anchor doesn't spin around in place.  The mount point for the anchor on the top_left_microswitch.stl is flush with the rest of the top surface, so there was nothing to prevent the original anchor design from spinning in place.

Old design: [image]

Modified design: [image]

And here it is installed: [image]

grilparto, 2024-05-10:
Suppose attaching the STL file would also be useful.  Given the limits on filetypes that can be upload, anyone grabbing this will need to change the .txt extension to .stl.

[cable_anchor-nema17offset.txt](https://github.com/jon-harper/E34M1/files/15279294/cable_anchor-nema17offset.txt)

jon-harper, 2024-05-14:
This is very cool work! Thank you for both reporting this *and* solving it.