jon-harper/E34M1 #42: User contribution:  ABL Side Mounts for BTT Microprobe

wagnerism, 2024-10-15:
I'm offering my Standard and UHF mounts for the BTT MicroProbe in STL format. I have personally tested both files with both Rapido HF and Rapido UHF hot ends.

I'd create a PR but I don't know how to create pictures consistent with what you have already.

I intentionally put the wire on the outside because I prefer a smaller probe X offset over hiding the wire and making it difficult to connect. 

[abl_btt_microprobe_standard_and_uhf.zip](https://github.com/user-attachments/files/17381986/abl_btt_microprobe_standard_and_uhf.zip)

jon-harper, 2024-12-30:
Thanks! I'll get to work on this. Apologies for the delay.

wagnerism, 2025-02-06:
No worries.  Thank you for your work here. I'm glad to contribute in some small way.

3D-Experiments, 2025-03-02:
The UHF version seems to be the right size for my set up. I'll post some results after testing. In the meantime, I've adjusted the orientation of the STLs to obviate supports. 

[abl_btt_microprobe_standard_and_uhf-NO-SUPPORTS-ORIENTATION.zip](https://github.com/user-attachments/files/19041228/abl_btt_microprobe_standard_and_uhf-NO-SUPPORTS-ORIENTATION.zip)

3D-Experiments, 2025-03-02:
Mounted height looks correct for my set up...

[image]

jon-harper, 2025-03-17:
This is really good stuff. Thank you both. What kind of hardware does this need? 2 inserts and 4x M3x6mm?

3D-Experiments, 2025-03-18:
M3X6 worked for me but I'll defer to @wagnerism to confirm.