jon-harper/E34M1 #44: TriangleLabs TD6S hotend mount. STL and STEP enclosed

53Aries, 2024-12-30:
Just figured I'd dump this here if you want to add it. 

[HOTEND_TD6S.zip](https://github.com/user-attachments/files/18273406/HOTEND_TD6S.zip)
[image]

jon-harper, 2025-03-05:
Checking that the values below are correct:

[image]

53Aries, 2025-03-05:
> Checking that the values below are correct:
> 
> [image]

Looks correct to me. Here's my full assembly that I've been using if you need to check anything. Thanks

[TD6S Assembly.zip](https://github.com/user-attachments/files/19095259/TD6S.Assembly.zip)

jon-harper, 2025-03-05:
I've pushed an update to the docs including the TD6S. Thanks for contributing. :-)