jon-harper/E34M1 #46: kraken_klicky very minimal clearance between klicky mount and printer bed

53Aries, 2024-12-31:
Just switched to kraken klicky from trihorn klicky and found out the klicky mount portion of kraken is waaaaay close to the bed. .155mm to be exact. Just figured if you still have the project file it might be easy to bump the klicky module up like 1mm. I'll work on it also. 
[image]
a

jon-harper, 2024-12-31:
what hotend is this?

jon-harper, 2024-12-31:
I'm not sure what the issue is with clearance. Are you concerned about weakening magnets?

53Aries, 2024-12-31:
> I'm not sure what the issue is with clearance. Are you concerned about weakening magnets?

It's probably just a me thing. I like having 2mm or more clearance just to avoid hitting curled perimeters, and to avoid my nozzle brush and purge bucket servo.

jon-harper, 2024-12-31:
That's fair enough. Let me see how hard the file will be to modify.

jon-harper, 2024-12-31:
[image]
[kraken_klicky_2mm.zip](https://github.com/user-attachments/files/18282985/kraken_klicky_2mm.zip)

Try this. 2mm offset to build plate.

53Aries, 2024-12-31:
Thanks. Can I get a .step to update my cad model?

jon-harper, 2024-12-31:
[kraken_klicky_2mm_step.zip](https://github.com/user-attachments/files/18283149/kraken_klicky_2mm_step.zip)

Not a problem. Let me know how this goes and I'll add it to the repo/docs.