jon-harper/E34M1 #51: Correct Printing Orientation?

3D-Experiments, 2025-02-28:
https://github.com/jon-harper/E34M1/blob/main/STL/top/top_left_microswitch.stl 

[image]

and

https://github.com/jon-harper/E34M1/blob/main/STL/top/top_right_microswitch.stl

[image]

have different printing orientations.

Does https://github.com/jon-harper/E34M1/blob/main/STL/top/top_right_microswitch.stl need to be flipped to avoid supports?

jon-harper, 2025-02-28:
> (jon-harper/E34M1#51) <https://github.com/jon-harper/E34M1/issues/51>

Ooh good find. Yes, that should be flipped.

jon-harper, 2025-03-05:
Fixed. Thanks.