jon-harper/E34M1 #53: E34M1 Beacon_Bottom_UHF Not Compatible with Beacon Low Profile Rev H

iblueup, 2025-04-13:
Beacon Rev H has components on the top of the PCB that result in a tilt of the scanner when mounted to the mentioned duct.

[image]

I temporarily resolved this by removing material post-printing. 

[image]

A fix for this is made for standard EVA 3, by shortening the mounting boss to clear SMDs, but this does not fit the E34M1 spacing between front and rear.
https://www.printables.com/model/865950-eva-3-beacon-revh-uhfvolcano

Any help to adjust the E34M1 duct to accommodate Rev H are appreciated!

jon-harper, 2025-04-13:
Thanks for the report. The easiest fix is to split the Rev D and Rev H models, I think.

iblueup, 2025-04-16:
Thanks for the attention on this! Looking at all the bottom duct options, I can imagine it'll take some effort. 

Really appreciate you making such a wonderful project available and continuing to support it!!

jon-harper, 2025-04-17:
By the way, you can use the Kraken UHF as-is with any Beacon rev, just update the Y offset.